speeduino/Hardware #1: 0.4.3d 22uf ceramic capacitor in reset protection

oleh-danilovskyi, 2021-11-01:
I can see that 0.4.3d has C19 22uf  in "Reset protection". It is 
FA22X7R1E226MRU06 and pretty expensive. According to AVR040 4.7nf should be used for reset pin protection . Actually previous version of speeduino  used 4.7nf instead of 22uf. Is there a need to have there such "big" capacitor?

turboedge, 2021-11-01:
If you really want it to prevent any resets, even ones caused by connecting the USB cable and other things, then the higher value is required. The lower values are just to prevent interference from causing a reset. There are other 22uf ceramic capacitors that might be cheaper, and probably more than fine.

noisymime, 2024-02-05:
Short answer, yes, the bigger cap is needed with the 2560. For some reason it is very sensitive to the capacitor value (Much more so than other AVR MCUs). 

As mentioned, other caps of similar values can certainly be used

closed 2024-02-05 by noisymime