speeduino/Hardware #3: 0.4.3d MAP sensor mounting hole blocked by Arduino USB connector

DeionSi, 2021-11-11:
[image]

tydelane, 2025-01-06:
Hi, did you find a solution to this? I am dealing with this issue right now

tydelane, 2025-01-07:
> I've been running my 4.3d like this for 2 years without issue. The second
> hole is not required.

Well thank you for letting me know, I'm just finishing up assembling my
0.4.3d and I'm going to install it with a k24A4 in my 98 civic! I had to
get a replacement for the MAP sensor though since the recommended was
obsolete. I chose the MPX5700. Was the 4250 available when you built yours
or did you have to get a replacement as well?

kendoman96, 2025-01-07:
I didn't have to get a replacement. 2 years later turbo drift miata has issues with everything except the ECU lol