speeduino/Hardware #4: 0.4.3d (and others?) Hardware SPI pins used by non SPI

DeionSi, 2021-11-28:
It was noted that atmega2560 hardware SPI (PB0-3) pins (arduino mega D50-D53) are used by non SPI functions (clutch input, ignition 3 output, ignition 4 output). Perhaps these circuits could be moved to other outputs to not restrict SPI usage.

noisymime, 2021-11-28:
What's the actual use case here as there's no SPI support for things with these boards? 

The v0.3/0.v4 can't and won't be changed on these pins. Any change here would need to be a whole new board.

DeionSi, 2021-11-30:
Actual possible use case would be for SD-card logging with something like https://electropeak.com/micro-sd-tf-card-adapter-module.
Even if not possible 0.3 or 0.4, this issue could be kept open as a reminder for future revisions. It is your choice.