speeduino/Hardware #15: Gerber outline (gml/gko/gbr) file mising

bilbolodz, 2024-02-01:
In project gerber Board Outline (gml/gko/gbr) file is missing. Could you please provide these?

turboedge, 2024-02-01:
Which board is it missing for?

bilbolodz, 2024-02-01:
I've tried to put your gerber for VR Conditioner on DirtyPCBs and it complains about missing Board Outline
I also tried with:

- THT v0_4_3d
- SMD v0_4_4d
[image]

turboedge, 2024-02-01:
Dirty PCB is not detecting it properly. It is the file edge cuts or contour, depending on which board you are working with. That VR conditioner was uploaded by someone else, and does seem to be missing the outside edge. I would use the v3.1 instead.

bilbolodz, 2024-02-01:
VR v3.1 has problem too. Do you have contact to person responsible for VR conditioner board?

bilbolodz, 2024-02-01:
It's probably: https://github.com/TechColab/VR-Conditioner-MAX9926-reg

turboedge, 2024-02-01:
Just uploaded this to another supplier and it worked perfectly.

[(https://github.com/speeduino/Hardware/blob/5883efc2f211134b859203c3047a4003eb73300a/VR%20Conditioner/Prior%20Versions/v3/gerbers/VR%20Conditioner-v0.3.1.zip)]