speeduino/Hardware #19: Power Supply Component Mismatch: Suggested Replacements

OpenLogicEFI, 2024-09-28:
Noticed a few components need to be balanced in the power supply. I found some component that could serve as BOM drop in replacements. 

**D15 has a max Vbr of 49.1v. Too high for D14 (30v) and U8 (35v).** 
-Could swap D15 to SMBJ20A-Q. Vbr is 24.5v.

**C21 doesn't meet the ESR requirements of U8 ([page 15 - 8.1](https://www.ti.com/lit/ds/symlink/lm2940-n.pdf?HQS=dis-dk-null-digikeymode-dsf-pf-null-wwe&ts=1727472433869&ref_url=https%253A%252F%252Fwww.ti.com%252Fgeneral%252Fdocs%252Fsuppproductinfo.tsp%253FdistId%253D10%2526gotoUrl%253Dhttps%253A%252F%252Fwww.ti.com%252Flit%252Fgpn%252Flm2940-n)), which calls for a 'critical' use of 100mΩ - 1Ω ESR.**

-Could replace U8 to make it less strict (TLF80511TC / TLE42744GV50ATMA, 400mA max, Automotive) or (LM7805SX/NOPB / LM340S-5.0/NOPB, higher output, easier to get, reliable).

-Or, replace C21 with something like TPSB476K010R0250 (250mΩ) or TPSB476K010R0350 (350mΩ).

noisymime, 2024-09-28:
Not sure if I'm looking at a different data sheet or BOM, but the ESR I'm seeing of C21 (EEU-FC1H470B) is 600mΩ, which is very much in spec?

OpenLogicEFI, 2024-09-28:
I think you're looking the 0.4.3d THT BOM. I was looking at the 0.4.4d SMD board.  

0.4.4d (https://github.com/speeduino/Hardware/blob/main/v0.4/SMD/Latest/v0.4.4d%20BOM.csv)