speeduino/Hardware #20: Missing v0.4 SMD symbols after importing in KiCAD 8

Thrathra, 2024-11-16:
Hi, I'm currently working on the v0.4.4d SMD project.

It seems that some symbols are lacking after importing the project.
For instance, a diode is missing at the right of the power regulator page 4, and the circuit at the right of the S1 component is missing aswell on the same page.
Do I have to install additional libraries and if so, which ones?
Furthermore, it is not clear if I have to use the "v0.4.4d.kicad_pro" file or the "v0.4.4d.pro" as they have different sizes.

Thanks in advance!

turboedge, 2024-11-21:
I'd have to check that, but I can say that you should be using the .kicad_pro file. It will be the newer of the two. The .pro files were used with even older versions of the software, and Kicad does not remove them when the project is upgraded.

Thrathra, 2024-12-02:
Thanks for your response and sorry for the delay.
Here's a screenshot about the issue.
[image]
Do you have the same schematic or is an issue on my side?